ffenix113/zigbee_home #9: Add wiki for project documentation?

Hedda, 2024-01-19:
Maybe consider adding wiki to this GitHub repository for project documentation? 

https://docs.github.com/en/communities/documenting-your-project-with-wikis

https://docs.github.com/en/communities/documenting-your-project-with-wikis/about-wikis

https://docs.github.com/en/communities/documenting-your-project-with-wikis/adding-or-editing-wiki-pages

A few pros and cons, but the benefit over Git based documentation is that it makes it easier to contribute to end-user documentation.

You could choose to do regular dumps from wiki to files that can be pushed to git if want a Git copy of the project documentation.

ffenix113, 2024-01-20:
Hello Hedda,

Thank you for the proposal. I also think that documentation is an important part of any project.
I thought to start with Github pages, which would have a benefit of git versioned, but also a benefit of web-pages(styling, navigation, etc).

I will add this to a list of things to work on in near future.

psi-4ward, 2024-01-23:
I could provide a vuepress scaffold if you like similar to zigbee2mqtt.io

vdbrink, 2024-01-23:
GitHub pages are a fast way to setup docs without spending time for a design.
On https://pages.github.com/themes/ you can find the supported basic themes for github pages.
Just follow the steps on https://docs.github.com/en/pages/getting-started-with-github-pages/creating-a-github-pages-site I was up and running in minutes. 

Here you can see my config file (from my small blog site) https://github.com/vdbrink/vdbrink.github.io/blob/main/_config.yml as example with some custom overrides via the _includes and _layouts directory.

I hope one of the first thing you document is how to define the custom component in Z2M.
This looks for me as a promising project!

ffenix113, 2024-01-24:
Thank you for suggestions.

For now I went with MKDocs & MKDocs Material. I think that this might change in the future, but for start that should suffice.

> I hope one of the first thing you document is how to define the custom component in Z2M.

@vdbrink I will add info about the configuration and adding sensors tomorrow, as it might take some time.

For now bare-bones docs site should be available here: https://ffenix113.github.io/zigbee_home/. Adding it here for reference, not as completely ready docs site.

Please note that this project is still in early development, so not everything might be ready now. Still, if you have any improvements for docs/project - please add a ticket or discussion for it.

Thank you for the support!

ffenix113, 2024-01-24:
I have updated documentation to provide examples, configuration options and CLI commands.

If there would be something missing or unclear - please feel free to open up another issue!

Hedda, 2024-01-25:
If wiki was not added then this should either be left open or closed as ”will not implement”. Not closed as completed if no wiki 😜

ffenix113, 2024-01-25:
@Hedda, fair enough. I thought that idea was to have a git-versioned documentation, and that Wiki is an option to do that.

Do you feel that the documentation on the site mentioned above is enough, or you would still suggest to have a Wiki in the project? For now I will close it as "not planned" to move this forward, but I am open for suggestion.

P.S.: @Hedda if you would like to - you can leave me some message and your contact info [here](https://airtable.com/appykiuiUYh56G6qD/shrDJxMEnnbmAZhIn), maybe we can talk some more about other things.

closed 2024-01-25 by ffenix113

Hedda, 2024-01-25:
Personally I like if there is also a Wiki regardless if there is a Git-versioned documentation because you are much more likely to get more people from the community to contribute to a Wiki then you can get submit PRs to Git documentation.

So yes it is great to have a Git-versioned documentation, but for general stuff that do not need versioning it good to also have a Wiki so that those that do not contributre to Wiki can still contribute via Wiki as long as it is not version specific.

ffenix113, 2024-01-25:
I understand now, thank you! Yes, it is a good idea to have it in this case!

I enabled Wiki and removed restriction on editing. Currently I am focusing on the code(but anybody else feel free to edit, if you would like), but will also add some information to Wiki in next days.