ffenix113/zigbee_home #17: Support for other sensors

lucaci32u4, 2024-01-29:
Hi,
I find your project really useful for building custom home automation on zigbee.

I'd like to contribute by implementing more sensor/actuator support. I'm especially interested in rotary encoders as sensors and light outputs. Any pointers where I should start to look to implement these?

Hedda, 2024-01-30:
Could sensor components be ported from ESPHome?

https://esphome.io/

https://github.com/esphome/esphome

Also check out the code for the projects linked in this discussion:

https://github.com/ffenix113/zigbee_home/discussions/15

ffenix113, 2024-01-31:
For near future this will be the doc on supporting new sensors: https://ffenix113.github.io/zigbee_home/sensors/supporting_new_sensors/

It is not complete, as I want to try and describe as much as possible related to supporting sensors, so keep an eye on updates.