ffenix113/zigbee_home #24: Monitor for Lithium battery cells

ffenix113, 2024-02-05:
### Discussed in https://github.com/ffenix113/zigbee_home/discussions/22

<div type='discussions-op-text'>

<sup>Originally posted by **Hedda** February  2, 2024</sup>
https://github.com/osaether/nrf_cell_mon</div>

Hedda, 2024-03-18:
While not sure if this "as-is" code is open source the Zigbee battery monitor concepts described here might still be useful:

https://github.com/SiliconLabs/zigbee_applications/tree/master/zigbee_battery_monitor

ffenix113, 2024-03-31:
I just merged ADC implementation with battery power and soil moisture ADC sensors, and I think this should be now supported, if configured correctly.

I will try this out when I would have ability and option to do so, or if anybody else would do this before I would have time for it - also great!