ffenix113/zigbee_home #35: Include a few Zigbee examples as reference sample applications/projects?

ffenix113, 2024-03-20:
### Discussed in https://github.com/ffenix113/zigbee_home/discussions/34

<div type='discussions-op-text'>

<sup>Originally posted by **Hedda** March 18, 2024</sup>
An idea could be to include a few Zigbee examples as reference application samples similar most commercial Zigbee SDKs, and since some of those samples are open souce it might be a good idea to copy and/or adapt those examples for reference that help with ideas. Check out:

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples

https://github.com/SiliconLabs/zigbee_applications

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/zigbee.html

https://software-dl.ti.com/simplelink/esd/plugins/simplelink_zigbee_sdk_plugin/1.60.01.09/exports/docs/zigbee_user_guide/html/zigbee/developing_zigbee_applications/example_applications.html

</div>

Hedda, 2024-04-10:
Good basic getting started Zigbee Home Automation (Zigbee HA) samples and a motion sensor example similar to these:

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample

https://github.com/SiliconLabs/zigbee_applications/tree/master/zigbee_smart_lighting

ffenix113, 2024-04-18:
Hello @Hedda ,

I've just added some examples via https://github.com/ffenix113/zigbee_home/pull/50, and will work on adding some examples from your comments as well.

For now let's keep this issue open until there are better examples to show more features, but do know that examples already are present.

Thank you!