ffenix113/zigbee_home #47: Considered using nrf-docker for builds?

evanraalte, 2024-04-16:
First of all, awesome project! I thought of this several years ago, but never started on it. Amazing to see that you took the initiative, thank you. 

Just listing this here, in case you hadn't seen it yourself, but there is a docker image that could do the firmware builds for you. This is very beneficial when you want to run tests, or make it possible for users to compile without setting up the whole toolchain. 

I might try to add this myself, but my main experience is in Python/C, so working with Go might be a bit of a hurdle here.

https://github.com/NordicPlayground/nrf-docker

evanraalte, 2024-04-16:
For instance:

```
 docker run --rm -v ${PWD}/firmware:/workdir/project -w /workdir/project nordicplayground/nrfconnect-sdk:main \ west build -p always -b nrf52840dk_nrf52840   
 
 ```

ffenix113, 2024-04-17:
Hello @evanraalte !

Yeah, I saw that docker image, as I researched options a bit for building example firmwares on tags, for example.

The issue to build example firmwares: https://github.com/ffenix113/zigbee_home/issues/36, workflow file that I used to test it with: https://github.com/ffenix113/zigbee_home/blob/feat/build-example-firmware/.github/workflows/example_build.yml

Building the firmware takes some time, and I don't want to use shared Github runners to build example firmwares(at least often, like on merge to `develop`), because it will eat free minutes quite quickly. Still looking for a good solution for this though!

As for adding ability for users to use this image - good idea, probably the option could be added to the build command to run builds inside the docker container, it would definitely save hassle of setting SDK yourself.

Thank you.

evanraalte, 2024-04-17:
@ffenix113 Ah yes, it is quite time consuming indeed. I'd say building these in CI on a release/merge to dev is probably still good to have. But perhaps, one could use a local test suite in addition, when developing new features? I wouldn't start with local runners on a public project, as that is a potential security issue.

Regarding the build option for users. Perhaps a `--build local` and `--build docker` would suffice? In that case a seperate `--flash` option is required as well I think.

ffenix113, 2024-04-23:
> But perhaps, one could use a local test suite in addition, when developing new features?

Currently I set up two toolchains through VSCode, but having option to run in Docker is a lot cleaner. Something to add, it would be a good addition.

> I wouldn't start with local runners on a public project, as that is a potential security issue.

I agree, and so does [Github](https://docs.github.com/en/actions/hosting-your-own-runners/managing-self-hosted-runners/about-self-hosted-runners#self-hosted-runner-security). I will set up Gihtub-hosted runner for firmware building and will see how it goes.

felipejfc, 2024-07-15:
Take a look https://github.com/ffenix113/zigbee_home/pull/73