ffenix113/zigbee_home #55: Code does not compile for more than 4 sensors

bbadrignans, 2024-04-30:
I got this error message when having 2 on_off, ias_zone, temperature and humidity. 
```
main.go:30: generate base: write app src: write template: execute source template "clusters.h.tpl": template: clusters.h.tpl:103:10: executing "clusters.h.tpl" at <$i>: wrong type for value; expected int; got *base.CommonDeviceClusters
exit status 1
```
Seems that for more than 4 end points, Nordic zboss force some differences, but I don't understand what [here ](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/zboss/3.11.2.0/using_zigbee__z_c_l.html#zigbee_device_context_mult_ep_dec).

Nevertheless, there is something wrong in templates/src/zigbee/clusters.h.tpl but don't know how to fix it.

ffenix113, 2024-05-10:
Thank you for the issue!

Good catch, I will check and fix this.

ffenix113, 2024-05-11:
The issue should now be resolved. If it is not - please feel free to re-open the issue.

closed 2024-05-11 by ffenix113

bbadrignans, 2024-05-11:
Yes great, thanks.
Tested with 1 dht and 4 on_off.

commit 62e0b104 (2024-05-11): fix: proper definition for device ctx with >4 endpoints

Closes #55