ffenix113/zigbee_home #70: Can't bulid firmware on Windows/Powershell

felipejfc, 2024-06-18:
Is building the firmware on powershell supposed to work? I get the error that template is not defined on all examples:

`go run .\cmd\zigbee\main.go  --config .\examples\factory_reset\zigbee.yaml firmware build
2024/06/18 12:39:43 generator.go:106: Device: "nrf52840dongle_nrf52840", selected bootloader: "nrf52_legacy", forced bootloader: false
2024/06/18 12:39:43 main.go:46: generate base: write app src: write template: execute source template "main.c.tpl": template: main.c.tpl:34:4: executing "main.c.tpl" at <maybeRenderExtender .Template "top_level" (sensorCtx 0 $.Device nil .)>: error calling maybeRenderExtender: extender template "peripherals/buttons" is not defined
exit status 1`

`go run .\cmd\zigbee\main.go  --config .\examples\battery_measurement\zigbee.yaml firmware build
2024/06/18 12:40:52 generator.go:106: Device: "arduino_nano_33_ble", selected bootloader: "arduino", forced bootloader: false
2024/06/18 12:40:52 main.go:46: generate base: write app src: write template: execute source template "main.c.tpl": template: main.c.tpl:34:4: executing "main.c.tpl" at <maybeRenderExtender .Template "top_level" (sensorCtx 0 $.Device nil .)>: error calling maybeRenderExtender: extender template "peripherals/adc" is not defined
exit status 1`

felipejfc, 2024-06-18:
Tried on WSL/Linux and on it I get this other error:

`go run ../../cmd/zigbee/main.go --config zigbee.yaml firmware build
2024/06/18 12:55:01 generator.go:106: Device: "xiao_ble", selected bootloader: "adafruit_nrf52_sd140_v7", forced bootloader: true
2024/06/18 12:55:01 ncs_finder.go:75: requested toolchain version: "v2.5.0", available versions: [v2.6.1 v2.5.0]
2024/06/18 12:55:01 device.go:156: found toolchain version "v2.5.0", requested version "v2.5.0"
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V]
            <command> ...
west: unknown command "build"; do you need to run this inside a workspace?
2024/06/18 12:55:02 main.go:46: build firmware: run command "west": exit status 1
exit status 1`

generation of project worked fine but west is not recognizing the build command

ffenix113, 2024-06-19:
Hello @felipejfc,

I didn't try to use it on Windows, but I can guess how to solve the problem in your original message.
For the second one - probably caused by [this single delete](https://github.com/ffenix113/zigbee_home/pull/64/files#diff-7ebe6ac6c28cb0e9cf877a4a61e8006d7785e08c645f7b93e826ab50bd8a475fL130).

Thank you for reporting this, will work on it.

ffenix113, 2024-06-19:
I have just merged some changes related to this.

Unfortunately, it will not work on Windows for now, as some paths are hardcoded. A good issue to work on.

Though on WSL it probably should work, if nRF SDK is installed as VSCode would(see [this](https://ffenix113.github.io/zigbee_home/using_the_cli/building_the_firmware/#preparing-environment)). 

If not - please tell how the SDK was installed and where it is located(do you build other firmwares with this install, or only use it for this project?).