ffenix113/zigbee_home #78: undefined node label 'btn1'

remy, 2024-10-27:
I'm trying to follow the initial example from the docs, but I can't get to the built firmware.

I get the following error after the build is configured:

```
-- Board: nrf52840dongle, qualifiers: nrf52840
-- Found host-tools: zephyr 0.16.5 (/opt/nordic/ncs/toolchains/f8037e9b83/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.5 (/opt/nordic/ncs/toolchains/f8037e9b83/opt/zephyr-sdk)
-- Found Dtc: /opt/nordic/ncs/toolchains/f8037e9b83/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6") 
-- Found BOARD.dts: /opt/nordic/ncs/v2.7.0/zephyr/boards/nordic/nrf52840dongle/nrf52840dongle_nrf52840.dts
-- Found devicetree overlay: app.overlay
devicetree error: /aliases: undefined node label 'btn1'
```

I'm using a supermini nrf52840 and wonder if there's a problem with having selected `nrf52840dongle/nrf52840` (I note there's no button on my supermini, though there's also only oe button on the dongle version… and the code/yaml refers to btn2 also).

Any ideas - happy to provide more examples, my `zigbee.yaml` is directly from here: https://github.com/ffenix113/zigbee_home/blob/develop/zigbee.yaml

ffenix113, 2024-10-27:
Thank you and sorry for the issue!

I will check what is going wrong there.

ffenix113, 2024-10-27:
Latest develop will contain a fixed version of the configuration.

Thanks for reporting this!

commit 1f2cdef8 (2024-10-27): fix: wrong button label in main example

Button labels without pins defined
must be references to buttons already defined
in board configuration.

Closes #78
PR #79 (2024-10-27): fix: wrong button label in main example
Button labels without pins defined
must be references to buttons already defined
in board configuration.

Closes #78
commit 17bb7b9e (2024-10-27): fix: wrong button label in main example (#79)

Button labels without pins defined
must be references to buttons already defined
in board configuration.

Closes #78
closed 2024-10-27 by ffenix113